Matthew-Hsu/PiPass #95: Not working anymore

suzsuzuki, 2018-04-02:
It seems Nintendo shut down their server. I get no more streetpasses. 

RIP HomePass

ourovoros, 2018-04-03:
Do you know if there was an official announcement? Actually I was thinking the same thing. I've noticed that I have no streetpasses at all for several days ago, I reinstalled Linux and tried reinstalling PiPass, but it seems that I could not get streetpasses anymore. At first I thought that 3DS was broken, but the actual 3DS devices are communicating each other normally.

ourovoros, 2018-04-03:
Ah, Please disregard previous comments. I found press release on Nintendo official site. According to it (https://www.nintendo.co.jp/support/information/2018/0228.html), they'd shutdown streetpass relay server on March/28/2018 10:00 AM JST. so I'd like to say thank you for PiPass.

closed 2024-08-31 by Matthew-Hsu